once-for-auto-driving/once_devkit #2: How to get the 2D box of object？

kai4729, 2021-07-05:
How to get the 2D box of obj? 
I can not get the correct 2D box from annotation file，Can someone help? Thanks a lot.

for example:
annotation in file "3D_infos\train_infos\data\000080\000080.json"   

> 		"sequence_id": "000080",
> 			"frame_id": "1616348814299",
>                           ...
>                          "boxes_2d": {
>                          ...
>                            "cam03": [
> 						[
> 							1231.188,
> 							548.4658,
> 							1689.1211,
> 							843.34955
> 						],
> 						[
> 							-1.0,
> 							-1.0,
> 							-1.0,
> 							-1.0
> 						],
> 						[
> 							1035.4642,
> 							539.30255,
> 							1114.8275,
> 							600.7152
> 						],
> 

draw the 2D box in image 3D_images\train_cam03\data\000080\cam03\1616348814299.jpg, python code and image result like this
`    cv2.rectangle(img, (1231, 548), (1689, 843), (255,0,0), 2)    `
`     cv2.rectangle(img, (1035, 539), (1114, 600), (255,0,0), 2)   `

[image]
the 2D box seem have some offset.

nmzfrank, 2021-07-05:
We only annotate the 3D bounding boxes. 2D boxes are the projection of 3D bounding boxes.

closed 2021-07-05 by nmzfrank

qingzi02010, 2021-08-10:
I have  2 confusions about the 2d and 3d boxes in 'val' dataset.
1. In seq_000028, cam01_1616102534801,  one of the results 'boxes_2d' is [0.0, 48.364197,1920.0, 1020.0], which is almost the full area and thus obviously wrong. I find many similar cases like that.
2. In many cases, the 'boxes_2d' result is -1, but the object is obviously visible. 
Considering the above 2 confusions, I am confused whether the 'boxes_2d' result is usable.