once-for-auto-driving/once_devkit #5: the Lidar and cameras install  pose

lgw1234, 2021-07-13:
I find the Data Annotation do not supply the Lidar and cameras's pose which are relative to self-Vehicle Coordinnate system,can you supply their install pose  ?

nmzfrank, 2021-07-14:
Vehicle coordinate system is same as the lidar coordinate system here in our system and camera-to-vehicle coordinate transform is the same as camera-to-lidar coordinate transformation.

lgw1234, 2021-07-15:
Thanks,can you provide  camera to-vehicle   coordinate transform when  the Vehicle coordinate system is defined at the bottom-center of  Vehicle-head ,not at the  lidar coordinate system ? beacue I want to  measure distance by Monocula front camer,such as camer-3, Finally, I can eval my precision of measure-distance by comparing lidar

nmzfrank, 2021-07-16:
As far as I known,  monocular depth prediction predicts the distance of objects by giving distances to pixels in the image. In this case the predicted distance can be defined as the distance between objects and camra/lidar/bottom-center of Vehicle-head/etc.  as long as the defination does not change throughout training and testing. So I can't figure out the necessary to provide such transformation. If there do exist some particular reasons, it is welcome to give more information.

closed 2021-07-21 by once-for-auto-driving