once-for-auto-driving/once_devkit #6: I can't download raw data using command line

Len-Li, 2021-07-16:
Hi, i try to download raw dataset using gdown. However, errors occur:

Access denied with the following error:

        Too many users have viewed or downloaded this file recently. Please
        try accessing the file again later. If the file you are trying to
        access is particularly large or is shared with many people, it may
        take up to 24 hours to be able to view or download the file. If you
        still can't access a file after 24 hours, contact your domain
        administrator.

You may still be able to access the file from the browser:

         https://drive.google.com/uc?id=1Yn4epcPn1dldb4YW6yX0WrmTZKXmzrKI

So could you update raw data link? As for train/val/test data, i can successfully download it.
Thanks!

nmzfrank, 2021-07-16:
We update the download link on July 17, so are you downloading through the new download links?

nmzfrank, 2021-07-16:
We also check that the link you provide can still be downloaded through browser, so if you are not in a hurry you can try later or download by browser.

Len-Li, 2021-07-16:
> We also check that the link you provide can still be downloaded through browser, so if you are not in a hurry you can try later or download by browser.

I am using the new link. 
Browser download is worked for me too. However, I am working on server, so I have to use command line to download dataset. So for me, I am using`gdown https://drive.google.com/uc\?id\=1Yn4epcPn1dldb4YW6yX0WrmTZKXmzrKI` 
to download dataset. 
The above command line is ok for train/val/test, but not work for raw dataset.
In addition, before July 17, the above command line isn't work for the whole dataset(train/val/test/raw). After the update, train/val/test download is worked for me except raw data.

Divadi, 2021-07-18:
I was having the same issue.
I am able to download it by following the tips from
https://stackoverflow.com/questions/65312867/how-to-download-large-file-from-google-drive-from-terminal-gdown-doesnt-work

nmzfrank, 2021-07-19:
I am not very clear about the download mechanism of gdown tools, maybe it is some sort of flow control. @Divadi provides a solution to solve the problem and you can try if it works for you @Len-Li.

Len-Li, 2021-07-20:
> I was having the same issue.
> I am able to download it by following the tips from
> https://stackoverflow.com/questions/65312867/how-to-download-large-file-from-google-drive-from-terminal-gdown-doesnt-work

This method works for me. Thanks!

closed 2021-07-20 by Len-Li