once-for-auto-driving/once_devkit #8: The coordinate system of the point cloud is inconsistent with that given on the homepage of the website？

Fitz-liu, 2021-07-20:
The coordinate system of the point cloud seems to rotate 90 degrees to the left relative to the coordinate system given on the homepage of the website
[image]

Fitz-liu, 2021-07-20:
got it

Fitz-liu, 2021-07-20:
Does anyone have this issue?

nmzfrank, 2021-07-21:
Hello @Fitz-liu, the coordinate system in your visualization is right, which is: X axis to the left; Y axis to the back and Z axis to the top. It is our mistake that the illustration image of the sensor on homepage is inconsistent. We will fix the coordinate and correct it soon.

Fitz-liu, 2021-07-21:
So, at present, if I want to use the data, I just need to rotate the coordinate system and the coordinate of the marked data by 90 degrees, right？

PointsCoder, 2021-07-21:
The annotations of 3d boxes and point clouds are under the same coordinate system, so you can just use the original coordinates. Rotating it by 90 degrees is also fine following the KITTI coordinate system.

once-for-auto-driving, 2021-07-21:
There are some typos in the descriptions of the coordinate system. We will fix those typos in the paper and website asap.

Fitz-liu, 2021-07-21:
Ok, thanks!

closed 2021-07-21 by once-for-auto-driving