once-for-auto-driving/once_devkit #9: Odd mismatch b/n 2D bboxes and images

Divadi, 2021-07-20:
I have visualized some of the 2D bounding boxes onto the undistorted images, and I got some odd results:
[image]
The truck annotation on the left seems to be a bit mismatched, and zooming into the cyclists annotations, there doesn't seem to be anything there:
[image]

Some other curiosities:
There seems to be some annotations for objects completely occluded (left):
[image]
Offset boxes:
[image]

It almost seems like there is a temporal mismatch between annotations and 2D images (3D -> 2D projected boxes would be larger than tight 2D boxes but not be offset like this) - did I perhaps overlook some transformation/is there anything I can do to remedy this issue?

Divadi, 2021-07-21:
Okay, I changed `once.py`'s undistort image function:
```
    def undistort_image(self, seq_id, frame_id):
        img_list = []
        split_name = self._find_split_name(seq_id)
        frame_info = getattr(self, '{}_info'.format(split_name))[seq_id][frame_id]
        for cam_name in self.__class__.camera_names:
            img_buf = self.load_image(seq_id, frame_id, cam_name)
            cam_calib = frame_info['calib'][cam_name]
            h, w = img_buf.shape[:2]
            # Previous 
#             cv2.getOptimalNewCameraMatrix(cam_calib['cam_intrinsic'],
#                                           cam_calib['distortion'],
#                                           (w, h), alpha=0.0, newImgSize=(w, h))))
#             img_list.append(cv2.undistort(img_buf, cam_calib['cam_intrinsic'],
#                                           cam_calib['distortion'],
#                                           newCameraMatrix=cam_calib['cam_intrinsic']))
            # New
            cam_calib['new_cam_intrinsic'], _ = cv2.getOptimalNewCameraMatrix(cam_calib['cam_intrinsic'],
                                          cam_calib['distortion'],
                                          (w, h), alpha=0.0, newImgSize=(w, h))
            img_list.append(cv2.undistort(img_buf, cam_calib['cam_intrinsic'],
                                          cam_calib['distortion'],
                                          newCameraMatrix=cam_calib['new_cam_intrinsic']))
        return img_list
```
To use the new cam intrinsic for undistorting the image.
Doing so, I was able to get seemingly correct 2D boxes:
[image]
[image]
[image]

To check, I also wanted to project the 3D boxes myself. I projected using the new_cam_intrinsic matrix and they seem quite close:
Colored boxes are 3D, black boxes are 2D:
[image]
However, some boxes don't line up:
[image][image]

And for cam07 images, my projected boxes seem to be "lower" for all sequences:
[image]
[image]

I am hoping to get some guidance on this

once-for-auto-driving, 2021-07-21:
Thanks for your attention to our dataset!

Offset boxes: We tested the image undistortion and projection using OpenCV (4.1.1) and didn't find such an issue, so we guess it is a problem with different OpenCV versions. Could you please provide the environment you use?

Occluded boxes: Our Lidar sensor is installed at a higher position compared to the cameras, so some objects that the lidar sensor can "see" may be occluded on the images. We annotated 3D boxes in point clouds captured by the lidar sensor.

Missing/Odd boxes: There are still some annotation errors even after two rounds of humans' checks. We plan to give another check to the annotations asap. Considering we are holding a workshop and a challenge, we may update the annotations later in this/next year.

nmzfrank, 2021-07-21:
Please refer to new commit https://github.com/once-for-auto-driving/once_devkit/blob/b659f4d66b5a3dceeaae9b8c66ae441e9379a8a3/once.py#L188. Misalign on cam07 is due to wrong distortion process in your code.

Divadi, 2021-07-21:
Thank you for your prompt response!

> Thanks for your attention to our dataset!
>
> Offset boxes: We tested the image undistortion and projection using OpenCV (4.1.1) and didn't find such an issue, so we guess it is a problem with [...]

Offset boxes: I had been using 4.2, I changed to 4.1.1 and there was no difference
Occluded boxes: I understand! So they are visible to the lidar but not the camera

------------------------
> Please refer to new commit
> 
> https://github.com/once-for-auto-driving/once_devkit/blob/b659f4d66b5a3dceeaae9b8c66ae441e9379a8a3/once.py#L188
> 
> . Misalign on cam07 is due to wrong distortion process in your code.

Running `once.py` as is, I had to change a few things:
I only have train & val downloaded, so only looped over those here:
https://github.com/once-for-auto-driving/once_devkit/blob/b659f4d66b5a3dceeaae9b8c66ae441e9379a8a3/once.py#L100

It seems like in the json, frames with annotations have frame_id as a string value, and frames without annotations have it as an int value, which causes isuses here
https://github.com/once-for-auto-driving/once_devkit/blob/b659f4d66b5a3dceeaae9b8c66ae441e9379a8a3/once.py#L112
and when when sorting here:
https://github.com/once-for-auto-driving/once_devkit/blob/b659f4d66b5a3dceeaae9b8c66ae441e9379a8a3/once.py#L125
So, I just changed line 112 to 
```
frame_list.append(str(frame_anno['frame_id']))
```

Then I changed this as I think rectangle takes a start point & end point
https://github.com/once-for-auto-driving/once_devkit/blob/b659f4d66b5a3dceeaae9b8c66ae441e9379a8a3/once.py#L203
to
```
cv2.rectangle(img_buf, tuple(box2d[:2]), tuple(box2d[2:]), (255, 0, 0), 2)
```

------------------------
With this, for seq 000076, frame 1616343527200, cam03:
[image]

As I mentioned above, **it seems like in `undistort_image`, the call to `cv2.getOptimalNewCameraMatrix` seems to not be doing anything:**
https://github.com/once-for-auto-driving/once_devkit/blob/b659f4d66b5a3dceeaae9b8c66ae441e9379a8a3/once.py#L144-L157
I thought it should return a new camera matrix & roi, and the new camera matrix should be used for undistort like this (with roi ignored since alpha is 0.0):
```
def undistort_image(self, seq_id, frame_id):
        img_list = []
        split_name = self._find_split_name(seq_id)
        frame_info = getattr(self, '{}_info'.format(split_name))[seq_id][frame_id]
        for cam_name in self.__class__.camera_names:
            img_buf = self.load_image(seq_id, frame_id, cam_name)
            cam_calib = frame_info['calib'][cam_name]
            h, w = img_buf.shape[:2]
            
            cam_calib['new_cam_intrinsic'], _ = cv2.getOptimalNewCameraMatrix(cam_calib['cam_intrinsic'],
                                          cam_calib['distortion'],
                                          (w, h), alpha=0.0, newImgSize=(w, h))

            img_list.append(cv2.undistort(img_buf, cam_calib['cam_intrinsic'],
                                          cam_calib['distortion'],
                                          newCameraMatrix=cam_calib['new_cam_intrinsic']))
        return img_list
```
Doing so yields 
[image]

Please let me know if my corrections are wrong - even with these changes, I have the issue of misalignment b/n 3D & 2D boxes for some frames

nmzfrank, 2021-07-23:
Hello @Divadi , thanks for your suggestion. We find out that the 2D projection is different from 3D lidar projection onto images and update a new resort image method to solve the problem as well as the 'str' and 'coordinate' bug you mensioned before.

As we mentioned, the 2D bounding boxes is the projection from 3D bounding box and is generated by annotation system automatically with no human check. As a result some 2D boxes which is 'behind' the camera may be reserved, causing some odd bounding box projects. So currently it is not recommended to use these noisy 2D annotations for supervision signals.

Divadi, 2021-07-23:
Thank you for these changes, and thank you for your patience. I have one last error, and this is regarding cam07. (Again, sorry for this long comment, but I had to include visualizations)

I understand that 2D bounding boxes are generated by projecting 3D boxes and that there may be errors. However, the 3D projected boxes generally do include the object, albeit not tightly, and I am hoping to use this for additional supervision. I am hoping to project the 3D boxes and resolve some of the incomplete annotations myself.
At the very least, I am trying to setup projection from 3D to 2D for lidar.

Here are some cam07 outputs from the given `project_lidar_to_image` function:
`'000092', '1616442487300'`
[image]
`'000076', '1616343527200'`
[image]

I notice that there seems to be misalignment (I observe no such issue for other camera frames).
So, I made a small adjustment to only take points _within_ a 3D bbox annotation:
```
def project_lidar_to_image(self, seq_id, frame_id, only_pts_in_box=True):
        points = self.load_point_cloud(seq_id, frame_id)
        
        if only_pts_in_box:
            import torch
            from pcdet.ops.roiaware_pool3d.roiaware_pool3d_utils import points_in_boxes_cpu
            annos = self.get_frame_anno(seq_id, frame_id)
            mask = points_in_boxes_cpu(points[:, :3], torch.tensor(annos['boxes_3d']))
            mask = torch.any(mask, dim=0) # point is in any box
            points = points[mask]

       ... rest is same...
```
Doing so, the misalignment is clear:
[image]
[image]

Some other examples:
`'000076', '1616343844199'`
[image]
`'000092', '1616442892300'`
[image]
`'000092', '1616443070800'`
[image]
[image]
`'000092', '1616442955301'`
[image]

For other cameras, the alignment is extremely precise:
[image]
[image]
[image]
So I feel it is an issue with something about cam07.

However, I find that **the cam07 2D bbox labels are very accurate**, while my projected 3D boxes instead match the points-in-boxes (so mis-matched with 2D annotations). 
So, the methodology used to generate the 2D box annotations were probably correct, but is perhaps not what is used in `once.py`?
(please see last two images in my second comment above, and here are more samples):
black boxes are 2D annos in json, colored 3D boxes are boxes I projected, and only points in some 3d box are shown:
[image]
[image]

And to show that my script does not have bugs, I use the exact same script for other cameras:
[image]
[image]

Please let me know if I made an error/if there is something I can do from my side to rectify this cam07 mismatch.
I apologize for the long post

nmzfrank, 2021-07-26:
Hello @Divadi , thanks for your detailed analyze and feel free for the long post as long as you feel it necessary to provide the images. 
Actually the misalignment of cam07 comes from the movement of the vehicle. While Lidar rotates one round to collect one frame, images are token by cameras. Though we try to keep sync between camera and lidar, there is some time gap between the moment images are shot and lidar points are returned, which means some image are token 'after' lidar points. This is perticular obvious when the speed of vehicle is high or the vehicle is turning. Take '000092', '1616443070800' for example, actually if you check the other cameras of the same frame you can find out that the misalignment happens too.

As for the 3D bounding box and 2D box mismatch, currently we provide two undistort method for 3D and 2D seperatly. You should use undistort_image for 3D bounding box project and undistort_image_v2 for 2D box. So I think 3D and 2D box can not draw in the same image.

Divadi, 2021-07-26:
Really, thank you so much for your patience and efforts.

I see; that does make sense.
Visualizing that frame:
cam7,8,9:
[image]
[image]
[image]
8 seems alright, but I can see that in 9, the left-most pedestrian has some offset. Although it is a bit weird that cam07 seems to most consistently have this issue.

Lastly, beyond this issue, the 2D boxes were generated from 3D box projection without human intervention. So, there must be _some_ way 3D can accurately be projected onto cam07, right? Because I see that 2D boxes on cam07 are very good (no offset issue like there is for 3D), examples shown way above. I have tried 3D projection for both undistort_image and undistort_image_v2, as well as the old and new camera intrinsics (all four combinations of this), but the 3D and 2D do not line up for any of them. How were the high-quality cam07 2D annotations generated?

nmzfrank, 2021-08-05:
Hello @Divadi , firstly,  I'd like to thank you for your patient and careful check on ONCE dataset. 
It takes some time to re-check the rawdata and calibration information.  We find out that calibration info to the part of the data is wrong, which causes the misalignment of lidar projection for cam07. We update all the six info files (which contains all the calibration data) to baidu Yunpan and google drive. You can download the updated files through [ONCE dataset](https://once-for-auto-driving.github.io/download.html#downloads). 
We also correct the projection function for both 2D and 3D annotations in https://github.com/once-for-auto-driving/once_devkit/blob/c084b5d49a029e34570d4ddb7b7011052cb955ee/once.py#L212, which uses the new camera intrinsic matrix and now 2D and 3D boxes matches.

closed 2022-01-10 by nmzfrank