once-for-auto-driving/once_devkit #21: ego pose meaning

turboxin, 2021-09-26:
Hi there,

I notice that in your paper, pose means the transformation matrix between each two adjacent LiDAR coordinates
[image]

Hwever, in the documentation from [https://once-for-auto-driving.github.io/documentation.html#data_format](url), pose means relative pose to the first frame of the scene.
[image]

I'm wondering which one is correct?

turboxin, 2021-09-26:
And is if a problem with the order of pose,
[image]

(quat_w, quat_x, quat_y, quat_z, trans_x, trans_y, trans_z) seems wrong as it is against [https://github.com/once-for-auto-driving/once_devkit/blob/c084b5d49a029e34570d4ddb7b7011052cb955ee/once.py#L280](url),  
where from_quat takes quaternion in (x,y,z,w) format.

nmzfrank, 2021-09-27:
In ONCE dataset, the pose infomation is relative to the first frame of the scene. You can refer to  [frame_concat](https://github.com/once-for-auto-driving/once_devkit/blob/master/once.py#L258).

closed 2022-01-10 by nmzfrank