once-for-auto-driving/once_devkit #22: Incorrect documentation

tarashakhurana, 2021-10-10:
Could you please update the 'pose' documentation to say that the pose is given as `(x, y, z, w)` and not `(w, x, y, z)`? Took me 3 days to figure this out. Will save time for others.

SSLAD2021, 2021-10-13:
Please refer to https://github.com/once-for-auto-driving/once_devkit/blob/master/once.py#L280, the 'pose' is (w, x, y ,z) rather than (x, y, z, w).

tarashakhurana, 2021-10-13:
Rotation.from_quat() takes in (x, y, z, w) https://docs.scipy.org/doc/scipy/reference/generated/scipy.spatial.transform.Rotation.from_quat.html

SSLAD2021, 2021-10-13:
Got it. Mistaken the order of quat (x, y, z, w）with quat, trans_x, trans_y, trans_z. 
Documentation has been fixed and thank you for your correction.

tarashakhurana, 2021-11-02:
Thanks a lot for the quick fix! I also wanted to mention that you should call out in the documentation that the list of frames for every scene are not all at 0.5s interval, sometimes there is a gap of 1.5s - 7s between frames. I'm working on LiDAR forecasting (multi-sweep tasks) and so found this.

SSLAD2021, 2021-11-10:
Yes, the gap over 0.5s is caused by the frame drop during data collecting time when one or more cameras can not sync with lidar frame. We will update the documentation later.

closed 2022-01-10 by nmzfrank