once-for-auto-driving/once_devkit #26: 3d object detection test set

mrqrs, 2021-12-28:
Hi, What is the number of testset for 3d object detection? I generated the 15858 annos. And i submit to the evaluation web, it tells me that the number between gt and pred is not same.

nmzfrank, 2021-12-29:
Here is a [reference submission demo](https://drive.google.com/file/d/1mU81KoTxwTon8jzOrf6qxECaXfL5ID2z/view?usp=sharing) for test set, with contains the frame required to be evaluated.

mrqrs, 2021-12-29:
This demo contains 7897 frames. But the testing split data which is downloaded from this [web](https://once-for-auto-driving.github.io/download.html#downloads) contains more than 7897 frames. How do i select the corresponding data?

nmzfrank, 2021-12-29:
In the demo, each frame has its frame id, indicating the frame you should eval in your submission files.

mrqrs, 2021-12-29:
> In the demo, each frame has its frame id, indicating the frame you should eval in your submission files.

Ok, I will try it and resubmit result.

closed 2022-01-10 by nmzfrank