once-for-auto-driving/once_devkit #27: some evaluation (iou calculation) questions

gaojunbin, 2022-02-21:
@once-for-auto-driving 
Hi,

when ground truth box is [ 2.33053802e+01  2.28531830e+01 -1.19932727e+00  3.03450296e-01 2.71859976e-01  8.17803260e-01  4.71238898e+00] and prediction box is [ 2.33053820e+01  2.28531842e+01 -1.19932724e+00  3.03450286e-01 2.71859974e-01  8.17803264e-01  4.71241328e+00]. The IoU is -1.03465401e+00 calculated by you code while it should be about 1.0. I don't know if anyone meet this error or anything wrong I set.

Looking forward to your reply~

nmzfrank, 2022-03-04:
@gaojunbin 
I tested your box by evaluation code and the result is fine.
```
>>> import numpy as np
>>> from rotate_iou_cpu_eval import rotate_iou_cpu_one
>>> gt_box = np.array([ 2.33053802e+01, 2.28531830e+01, -1.19932727e+00, 3.03450296e-01, 2.71859976e-01, 8.17803260e-01, 4.71238898e+00])
>>> pred_box = np.array([2.33053820e+01, 2.28531842e+01, -1.19932724e+00, 3.03450286e-01, 2.71859974e-01, 8.17803264e-01, 4.71241328e+00])
>>> rotate_iou_cpu_one((gt_box, pred_box))
0.999970811510045
```

gaojunbin, 2022-03-04:
> @gaojunbin I tested your box by evaluation code and the result is fine.
>
> ```
> 0.999970811510045
> ```

Thanks. yeah, the result calculated by rotate_iou_cpu_one is right. But the code bellow can not get the right result.
```
gt_box = np.array([[ 2.33053802e+01, 2.28531830e+01, -1.19932727e+00, 3.03450296e-01, 2.71859976e-01, 8.17803260e-01, 4.71238898e+00]])
pred_box = np.array([[ 2.33053802e+01, 2.28531830e+01, -1.19932727e+00, 3.03450296e-01, 2.71859976e-01, 8.17803260e-01,  4.71241328e+00]])
print(iou3d_kernel_with_heading(gt_box, pred_box))
```
The function "iou3d_kernel_with_heading" is used in "get_evaluation_results" when I want to calculate the mAP for detection algorithm. Why does this function calculate iou and get abnormal results? Is there something wrong with my usage?

nmzfrank, 2022-03-04:
@gaojunbin 
Yes, you are right. 
This is caused by the annoying float point precision prolems, like below
```
(Pdb) B[1]
22.701462
(Pdb) A[1]
22.701456
(Pdb) B[1] - A[1]
5.722046e-06
```
This error is espacially sever for line intersection point calculation and result in wrong size of overlap area. 
It is fixed in commit [fix float point precision error](https://github.com/once-for-auto-driving/once_devkit/commit/49f4b723832be3210df69f365334bde2877c37db)
You can update your code by pulling the lastest code and the calculation should be consistent between GPU and CPU versions.

closed 2022-03-08 by nmzfrank